ROBOTIS-GIT/DynamixelSDK #577: Hi, was wondering if anyone as tried interfacing a Dynamixel Servo motor with the Universal Robot 5 ? If possible can please share how , any advice would be much appreciated. Thank you.

B3n77, 2022-07-30:
I am trying to interface a dynamixel servo in my gripper with a robotic arm(UR5), but unsure of how to interface my gripper. If possible is there any hardware recommended to connect the motor and the UR5? My motor is the XM540-W150R.

shiba-8ro, 2022-08-02:
Hi,
Unfortunately, Dynamixel does not support UR-CAP, which is UR's communication standard, so you cannot connect it to UR5.
You need to connect Dynamixel Hand to PC using USB interface such as U2D2.

Our robot hand RH-P12-RN-UR is compatible with UR-CAP.
https://emanual.robotis.com/docs/en/platform/rh_p12_rn_ur/

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon