ROBOTIS-GIT/DynamixelSDK #583: Unable to syncRead/bulkRead from different models

omeliailiffe, 2022-11-13:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  3.7.31

2. Which programming language/tool do you use?
  Java

3. Which operating system do you use?
  Windows 10

3. Which USB serial converter do you use?
  U2D2

4. Which DYNAMIXEL do you use?
  XM430-W250-R and XC330-T288-T

6. Please describe the issue in detail 

Attempting to read the hardware error from 9 motors (4 XM430, 5 XC330) using syncRead.
If I add all 9 to syncRead it give error `There is no status packet!`
If I change the code to only read the 4 XM430 or the 5 XC330 motors at once it produces the expected result.

Is syncRead from difference models not supported? 
I also tried bulkRead with the same outcome.

yun-goon, 2025-04-17:
Hello, and we apologize for the delayed response. When connecting DYNAMIXELs via TTL, we recommend a maximum of 15 units. The communication error "There is no status packet!" that you are encountering with a smaller number of DYNAMIXELs likely indicates a connection issue.

Please carefully check the cable connections between your DYNAMIXELs. Additionally, if possible, try replacing the USB cable connecting your U2D2 to your computer to rule out a faulty cable.

Thank you.

sunghowoo, 2025-06-20:
As there has been no recent activity, we will close this issue for now. If the issue still exists, please feel free to reopen it.

closed 2025-06-20 by sunghowoo