ROBOTIS-GIT/DynamixelSDK #587: where is the file :"dynamixel_functions_py"

lianuo, 2022-12-23:
in the emanual:

https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/python_read_write_protocol_1_0/#python-read-write-protocol-10

we have a sample code , but it cannot run with the SDK,some file are missing, how can I get the file dynamixel_functions_py?

ROBOTIS-Will, 2022-12-26:
Hi,

The example has been updated to install the python library, but the code in the documentation wasn't updated.
Please refer to the latest code in the below.

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol1_0/read_write.py

Thank you.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.