ROBOTIS-GIT/DynamixelSDK #588: Goal Current limits

ravitejanference, 2023-01-11:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
   v2.0.11.7

2. Which programming language/tool do you use?
   python

3. Which operating system do you use?
    ubuntu 18.04

3. Which USB serial converter do you use?
  `USB2Dynamixel

5. Which DYNAMIXEL do you use?
     XM430-W350

6. Have you searched the issue from the closed issue threads?
     yes
7. Please describe the issue in detail
    I want to increase the goal current(from 50 to 70) from my code but I am getting an error "goal current limit exceeded" this 
    is only for one system for other system I am able to increase the goal current  
8. How can we reproduce the issue?
    increase the goal current

shiba-8ro, 2023-01-20:
Hi, @ravitejanference 

You may need to change the Currect Limit of XM430 with Dynamixel Wizard2.0.
https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#current-limit38

Or PLZ try Dynamixel Firmware Recovery.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon