ROBOTIS-GIT/DynamixelSDK #594: Dynamixe MX 64 Firware Update Issue

attayyab, 2023-04-06:
Hello,
        I am using Dynamixel 64 and trying to update the firmware using Dynamixel wizard 2.0 while doing that it's not completing the update. The screen hang and both blue and green led of U2D2 blinking and red is solid.

shiba-8ro, 2023-04-10:
@attayyab 
Hi, Does the same thing happen even if you perform FW Recovery with the software Dynamixel Manager 2.0 that has the same function?
https://emanual.robotis.com/docs/en/software/rplus2/manager/

attayyab, 2023-04-11:
> @attayyab Hi, Does the same thing happen even if you perform FW Recovery with the software Dynamixel Manager 2.0 that has the same function? https://emanual.robotis.com/docs/en/software/rplus2/manager/

@shiba-8ro 
Hi,  I tried R+Manager also but while updating and testing it hangs at 57.14%.

[image]

shiba-8ro, 2023-04-11:
@attayyab 
Perhaps your mx-64's chip is faulty. Please take the RMA process from the place of purchase.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon