ROBOTIS-GIT/DynamixelSDK #597: @TW050203 @Huitongtong

EvgenRyabch123, 2023-05-27:
@TW050203 @Huitongtong 
The read_write_node.cpp example in the DYNAMIXEL SDK requires 2 DYNAMIXEL with ID 1 and 2.
If you are running with only one DYNAMIXEL (with ID = 1), please modify the source code to run with a single DYNAMIXEL.
For example, comment out the ID 2 torque enable code in line 131 as below
```cpp
  dxl_comm_result = packetHandler->write1ByteTxRx(
    portHandler, DXL1_ID, ADDR_TORQUE_ENABLE, 1, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS) {
    ROS_ERROR("Failed to enable torque for Dynamixel ID %d", DXL1_ID);
    return -1;
  }

  // dxl_comm_result = packetHandler->write1ByteTxRx(
  //   portHandler, DXL2_ID, ADDR_TORQUE_ENABLE, 1, &dxl_error);
  // if (dxl_comm_result != COMM_SUCCESS) {
  //   ROS_ERROR("Failed to enable torque for Dynamixel ID %d", DXL2_ID);
  //   return -1;
  // }
```

If you are running with AX series, you should modify the source code as the example is written for X series that have a different control table, data structure, and Protocol version from AX series.

Thanks.

_Originally posted by @ROBOTIS-Will in https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/558#issuecomment-1121768610_

EvgenRyabch123, 2023-05-27:
Hello! I have the same problem, I use Dynamixel for work : AH-12A.
Tell me what exactly needs to be changed. I don't understand. and where to find this information

yun-goon, 2025-02-14:
Hello @EvgenRyabch123! 
If you are using AX-12A, you need to modify the code because AX-Series uses a different control table and Protocol 1.0 compared to X-Series motors.

✅ 1. Change the Protocol Version to 1.0
In read_write_node.cpp, modify the protocol version:

```
#define PROTOCOL_VERSION 1.0  // Change from 2.0 to 1.0 for AX-12A
```
✅ 2. Update the Control Table Addresses
AX-12A uses different memory addresses than X-Series. Change these values:

Function | X-Series Address | AX-Series Address
-- | -- | --
Torque Enable | ADDR_TORQUE_ENABLE = 64 | ADDR_TORQUE_ENABLE = 24
Goal Position | ADDR_GOAL_POSITION = 116 | ADDR_GOAL_POSITION = 30
Present Position | ADDR_PRESENT_POSITION = 132 | ADDR_PRESENT_POSITION = 36

Modify your code like this:
```
#define ADDR_TORQUE_ENABLE 24
#define ADDR_GOAL_POSITION 30
#define ADDR_PRESENT_POSITION 36
```