ROBOTIS-GIT/DynamixelSDK #604: ROS2-Humble - Dynamixel Error: No module named ‘dynamixel_sdk_custom_interfaces’

alikhorshidi, 2023-08-20:
Hello!
I’m using ROS2 Humble to control 2XL430-W250-T connected via U2D2
I followed this Video-Tutorial:

[DYNAMIXEL Quick Start Guide for ROS 2](https://www.youtube.com/watch?v=E8XPqDjof4U)

Everything goes right until this command:

`ros2 topic pub /set_position dynamixel_sdk_custom_interfaces/msg/SetPosition `
after SetPosition, normally by pressing Tab, It should looks like in the video:

`ros2 topic pub /set_position dynamixel_sdk_custom_interfaces/msg/SetPosition "{id: 1, position: 1000}"`

but unfortunately, it’s not working!! and typing by myself is not solve the problem.

I also tried publishing this topic (set_position) using rqt but when I want to add, it causes crash!
crash title is:
[image]
ModuleNotFoundError: No module named ‘dynamixel_sdk_custom_interfaces’

There is “dynamixel_sdk_custom_interfaces” folder:

[image]

Please help me.

1. Which DYNAMIXEL SDK version do you use?
 Latest

2. Which programming language/tool do you use?
ROS2 - Humble

3. Which operating system do you use?
Ubuntu 22.04

3. Which USB serial converter do you use?
U2D2

4. Which DYNAMIXEL do you use?
2XL430-W250-T

@ROBOTIS-Will

ROBOTIS-Will, 2023-09-08:
@alikhorshidi 

I'm sorry about the delayed response.
Please see below forum thread for my test result and possible solutions.
https://forum.robotis.com/t/ros2-humble-dynamixel-error/4783/3?u=willson

yun-goon, 2025-02-14:
I think the problem has been solved, so I will close this issue.

closed 2025-02-14 by yun-goon