ROBOTIS-GIT/DynamixelSDK #606: NameError: name 'dynamixel' is not defined

MhmdBarazi, 2023-09-05:
Hello, I am using DYNAMIXEL SDK 3.7.51 with Python, on the Windows operating system. I am using U2D2 USB serial converter.
My motor is XC430-W240. I searched about this issue. I found some suggestions but none of them worked for me. 

I have oak camera as well as XC430-W240 motor. I want to have my robot move based on certain coordination given by the camera. Therefore, I extract output from the camera's function as an input to dynamixel function. Furthermore, I defined the movement function as a thread. and this is my code:

> import os
> import numpy as np
> from depthai_sdk import OakCamera, ArgsParser
> import argparse
> import depthai as dai
> from dynamixel_sdk import *
> from dynamixel_sdk.port_handler import PortHandler
> from dynamixel_sdk.packet_handler import PacketHandler
> from dynamixel_sdk.robotis_def import *
> import threading
> import queue
> 
> 
> parser = argparse.ArgumentParser()
> parser.add_argument("-conf", "--config", help="Trained YOLO json config path", default=r"C:\Users\user\Desktop\codes\320\best300.json", type=str)
> args = ArgsParser.parseArgs(parser)
> 
> 
> 
> 
> dx = 27.35
> dy = 123.0979
> dz = 45.57
> yT = 452
> x0 = 0
> y0 = 0
> z0 = 0
> Gamma = 0
> 
> 
> 
> 
> 
> MY_DXL = 'X_SERIES'       
> 
> 
> ADDR_TORQUE_ENABLE          = 64
> ADDR_LED_RED                = 65
> LEN_LED_RED                 = 1         # Data Byte Length
> ADDR_GOAL_POSITION          = 116
> LEN_GOAL_POSITION           = 4         # Data Byte Length
> ADDR_PRESENT_POSITION       = 132
> LEN_PRESENT_POSITION        = 4         # Data Byte Length
> DXL_MINIMUM_POSITION_VALUE  = 0         # Refer to the Minimum Position Limit of product eManual
> DXL_MAXIMUM_POSITION_VALUE  = 4095      # Refer to the Maximum Position Limit of product eManual
> BAUDRATE                    = 57600
> 
> 
> PROTOCOL_VERSION            = 2.0
> 
> 
> DXL_1_ID = 1;
> DXL_2_ID = 2;
> 
> 
> # Use the actual port assigned to the U2D2.
> # ex) Windows: "COM*", Linux: "/dev/ttyUSB*", Mac: "/dev/tty.usbserial-*"
> DEVICENAME                  = 'COM5'
> 
> TORQUE_ENABLE               = 1     # Value for enabling the torque
> TORQUE_DISABLE              = 0     # Value for disabling the torque
> DXL_MOVING_STATUS_THRESHOLD = 20    # Dynamixel moving status threshold
> 
> 
> 
> index = 0
> dxl_goal_position = 0        # Goal position
> dxl_led_value = [0x00, 0x01]                                                        # Dynamixel LED value for write
>         
> 
> 
> # Initialize PortHandler instance
> # Set the port path
> # Get methods and members of PortHandlerLinux or PortHandlerWindows
> portHandler = PortHandler(DEVICENAME)
> 
> # Initialize PacketHandler instance
> # Set the protocol version
> # Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
> packetHandler = PacketHandler(PROTOCOL_VERSION)
> 
> 
> # Open port
> if portHandler.openPort():
>     print("Succeeded to open the port")
> else:
>     print("Failed to open the port")
>     print("Press any key to terminate...")
>     getch()
>     quit()
> 
> 
> # Set port baudrate
> if portHandler.setBaudRate(BAUDRATE):
>     print("Succeeded to change the baudrate")
> else:
>     print("Failed to change the baudrate")
>     print("Press any key to terminate...")
>     getch()
>     quit()
> 
> # Enable Dynamixel Torque
> dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL_1_ID, ADDR_TORQUE_ENABLE, TORQUE_ENABLE)
> if dxl_comm_result != COMM_SUCCESS:
>     print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
> elif dxl_error != 0:
>     print("%s" % packetHandler.getRxPacketError(dxl_error))
> else:
>     print("Dynamixel has been successfully connected")
>     
>     
>     
>     
>     
> def GettingSpatial(packet):
> 
>         for det in packet.detections:
>             spatials = det.img_detection.spatialCoordinates
>             xc = spatials.x
>             x0 = xc + dx
>             zc = spatials.z
>             Alpha = np.arctan(yT/zc)
>             CosAlpha = np.cos(Alpha)
>             z0 = dz + (zc * CosAlpha)
>             Gamma = 10*np.rad2deg(np.arctan(x0/z0))
>             data_queue.put(Gamma)            
> 
> 
> 
> 
> def Movement():
>                 
>         while True:
>             # Send data
>             if not data_queue.empty():
>                 dxl_goal_position = int(data_queue.get())
>                 dynamixel.write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_1_ID, ADDR_PRO_GOAL_POSITION, dxl_goal_position[index])
>                 if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
>                     dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
>                 elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
>                     dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))    
> 
> 
> 
>         # Read present position
>                 dxl_present_position = dynamixel.read4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_1_ID, ADDR_PRO_PRESENT_POSITION)
>                 if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
>                     dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
>                 elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
>                     dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))
> 
> 
> 
> 
> 
> data_queue = queue.Queue()
> 
> # Start the Arduino thread
> movement_thread = threading.Thread(target=Movement)
> movement_thread.start()
> 
> 
> 
> 
>     
> with OakCamera(args=args) as oak:
>         stereo = oak.create_stereo(fps=8)
>         color = oak.create_camera('color', fps=8)
>         nn = oak.create_nn(args['config'], color, nn_type='yolo', spatial=stereo)
>         oak.visualize(nn, fps=True, scale=2/3)
>         oak.callback(nn.out.passthrough, GettingSpatial)
>         oak.start(blocking=True)    
>     
> # Disable Dynamixel Torque
> dynamixel.write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_1_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE)
> if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
>     dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
> elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
>     dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))
> 
> 
> # Close port
> dynamixel.closePort(port_num)

Issue #1:
everything worked perfectly before coordination detection( getting into Movement() function). then this error message has appeared: NameError: name 'dynamixel' is not defined
before getting into this function there is no such an error!
What is the source of this error?
Issue #2: 
I want to give the coordination in the degree unit as I did using Arduino. but I couldn't find the proper way.

Could you help me with them, please? 
Thank you!

yun-goon, 2025-04-17:
Hello @MhmdBarazi , I've reviewed your code and the issues you're encountering. Here are the explanations and solutions:

Issue 1: NameError: name 'dynamixel' is not defined

The reason for the NameError: name 'dynamixel' is not defined error is that you are trying to use a variable named dynamixel within your Movement() function without it being defined in that scope. While you import classes from the dynamixel_sdk, you haven't actually created an instance of a packet handler object and assigned it to the name dynamixel.

Solution: You need to instantiate your PacketHandler object and assign it to a variable that can be accessed within the Movement() function. You can either use the existing packetHandler object within the Movement() function directly or assign it to a new variable named dynamixel in the global scope.

Issue 2: Controlling the motor in degrees

Yes, your understanding is correct. The most straightforward way to control the DYNAMIXEL in terms of degrees, especially when you're aiming for coordinated movement based on camera input, is to work within the Position Control Mode.

### Explanation:

1. Obtain the desired angle: Your camera processing will provide a target angle.
2. Convert degrees to units: DYNAMIXELs in position control mode typically use a unit-based system for position values (e.g., 4096 units for 360 degrees for the XC430-W240). You need to convert your target angle from degrees to this corresponding unit value using the motor's specific resolution. The formula for XC430-W240 would be: Unit= 
360
Angle (degrees)×4096
​
3. Set the goal position in units: Use the calculated unit value as the goal position when communicating with the DYNAMIXEL using functions like write4ByteTxRx with the ADDR_GOAL_POSITION.
4. Read the present position (optional): If needed, you can read the current position of the motor, which will also be in units. You can convert this back to degrees for monitoring or further calculations if necessary, using the reverse of the conversion formula.
By following these steps, you can effectively control the DYNAMIXEL's position based on degree values derived from your camera input.

sunghowoo, 2025-06-20:
As there has been no recent activity, we will close this issue for now. If the issue still exists, please feel free to reopen it.

closed 2025-06-20 by sunghowoo