ROBOTIS-GIT/DynamixelSDK #608: Velocity control using ROS 2

Actat, 2023-10-24:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - latest foxy-devel (92f7035)

2. Which programming language/tool do you use?
  - `C++`, `ROS`
 
3. Which operating system do you use?
  - `Ubuntu 20.04`

3. Which USB serial converter do you use?
  - `U2D2`

4. Which DYNAMIXEL do you use?
  - `XM540-W150`

5. Have you searched the issue from the closed issue threads?
  - #127 implements velocity control using the older protocol.
  - #221 concerns how to obtain position information while performing velocity control.
  - #229 is about velocity control when using an Arduino.
 
6. Please describe the issue in detail
I want to use the dynamixel motor in velocity control mode. I need to get position information during control.
I was able to run the motor in position control mode from ROS 2 using read_write_node in dynamixel_sdk_examples. However, read_write_node does not support velocity control mode. Is there another node that supports velocity control mode?

7. How can we reproduce the issue?

shiba-8ro, 2023-10-30:
agreement +1

Actat, 2023-11-09:
I have found an unofficial way to solve my problem.
Is there an official way?

https://github.com/Actat/DynamixelSDK/commit/84d759e97b83343ecc9903ff4dd73599fd23e475

anasderkaoui, 2024-01-10:
Hello @Actat !
Can you please provide us with the full modifications that you did to the code.
I am at the same page, I want to use an MX-106 servo in velocity mode but I cannot get it done. I modified the code accordingly to yours @Actat 84d759e (ADDR_GOAL_**VELOCITY** 104 and id 1, addresses according to my servomotor) but the motor is still in position mode !
The command lines also seem to control position mode only: `ros2 run dynamixel_sdk_examples read_write_node`
returns Something like "[GOAL Position: 0]"
The other command also controls position : `ros2 topic pub -1 /set_position dynamixel_sdk_custom_interfaces/msg/SetPosition "{id: 12, position: 1000}"`
Is there another command that you used other than these ones above ?

yun-goon, 2025-07-08:
Hello!

For velocity control using ROS2, please try using https://github.com/ROBOTIS-GIT/dynamixel_hardware_interface. It has been developed to utilize ros2_control.