ROBOTIS-GIT/DynamixelSDK #612: Is it safe to upgrade MX-106T, MX-64T, MX-28T and AX-12A to protocol version 2?

AltziTS007, 2023-12-22:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

_Which programming language/tool do you use?_
  - **C, C++, Python**

_Which operating system do you use?_
  - **Windows 11, Ubuntu18.04**

_Which USB serial converter do you use?_
  - **U2D2**

_Which DYNAMIXEL do you use?_
  - **2xMX-106T, MX-64T, MX-28T and 2xAX-12A**

I have a robotic arm using the servos I mentioned above. Can I do a firmware recovery to those dynamixel servos without having any risk or problem in the future?  

Thank you!

shiba-8ro, 2023-12-25:
@AltziTS007 
You can update the MX series protocol version to 2.0 and safely revert to protocol 1.0 using Dynamixel Wizard 2.0. Important note. You need to connect Dynamixel and U2D2 one-to-one. FW recovery by daisy chain is not possible.
Unfortunately, the AX series does not currently support Protocol 2.0, and it is unlikely that it will support Protocol 2.0 in the future. In the current DYNAMIXEL series, only the AX series has an ATmega MCU. In order to use protocol 2.0 FW, the MCU must be ARM.

AltziTS007, 2023-12-29:
@shiba-8ro 
Thank you very much. 

Another thing that I would like to ask is if I can control this robotic arm with that dynamixel(protocol 1) that I mentioned using the dynamixel workbench.

shiba-8ro, 2024-01-08:
@AltziTS007 
I haven't used Workbench myself, but I know that it supports AX-12A and MX Series at Protocol1.0.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

You can use Workbench without worrying about the Protocol version since it's operations are pretty much wrapped. However, all servos must have the same Baudrate. Also, and this is the most important thing, every servo must have a different ID.

However, if you want to do more advanced development in the future, I highly recommend using the Dynamixel SDK.
My opinion is that Dynamixel Workbench is just a tutorial.

closed 2024-02-26 by AltziTS007