ROBOTIS-GIT/DynamixelSDK #661: Corrupt status message caused by negative numbers in extended position control mode

pvangoor, 2025-07-09:
Hi, I am using a xl330-M288 servo motor through python3 on Ubuntu 22.04.
I am able to connect to the motor through the wizard and have set it to "extended position control".
When I try to send a negative position through the python SDK (protocol 2.0), the motor reacts as expected (it moves to the expected position).
However, it also sends me a corrupt status message. This only happens with negative values for position. Positive values are always accepted.
An example corrupt message I receive is `rxpacket = [255, 255, 253, 0, 2, 4, 255]`.
I am sending the command using `write4ByteTxRx(self.portHandler, DXL_ID, ADDR_PRO_GOAL_POSITION, -500)`, with `ADDR_PRO_GOAL_POSITION = 116` and `DXL_ID = 2`. These match the values reported in the wizard.

Could you please help me resolve why I am getting corrupted status messages?
Please let me know if further detail is needed.

yun-goon, 2025-07-10:
Hi @pvangoor 

Have you updated the firmware of your XL330-M288 to the latest version?

Could you also please share the code you are using for reproduction?

We haven't encountered this issue when using the XL330-M288-T with the OpenRB.

Could you provide more detailed information on your setup, such as:

What specific microcontroller are you using? (e.g., Parallax Propeller - which model?)

What kind of "Dynamixel Bridge" are you using? (e.g., U2D2, specific custom board?)

How is the wiring configured between your microcontroller, the bridge, and the Dynamixel? (e.g., power supply, data lines, shared ground)

Are there any other devices on the same Dynamixel bus?

What is the exact firmware version of your XL330-M288 (e.g., 48 is listed, but any sub-version or build number)?

Any additional details about your hardware configuration and software environment would be very helpful in diagnosing this issue.

pvangoor, 2025-07-11:
Hi @Woojin-Crive,

I tried to set the return time delay to 0 but this made no difference to the results.

I also tried to set the baud rate to 4Mbps, but then I was no longer able to connect to the motor, and ended up needing to do a firmware recovery. Now it is back to the default baudrate of 57600, and I am getting the same results as described above.

Woojin-Crive, 2025-07-14:
@pvangoor Can you do the same test with [U2D2](https://emanual.robotis.com/docs/en/parts/interface/u2d2/)?